ussserrr/stm32pio #8: WARNING  STM32 PlatformIO board is not specified, it will be needed on PlatformIO project creation

kynguyen98, 2020-02-02:
Hi friend I was having trouble when following your guide and I can't find any solution to this
I follow your example section and when I type 
```python
python3 app.py new -d /home/john/STM32Cube/STM32F103C8/testing -b genericSTM32F103C8 --start-editor=code --with-build
```
It return me with this error
```
WARNING  [Errno 2] No such file or directory: 'platformio'
```
I created my platformio project in vscode with stm32cube framework and I put app.py inside the project and run the command again and it still give me this error 
Can you help me with this?

ussserrr, 2020-02-06:
Hello, seems like you don't have `platformio` command in your PATH, so `stm32pio` cannot start it. Please see [https://docs.platformio.org/en/latest/core.html](this) and [https://docs.platformio.org/en/latest/installation.html#piocore-install-shell-commands](this).

kynguyen98, 2020-02-13:
Okay I installed platformio and its in my PATH but I'm getting this error hope you could help me with 
```
ERROR    code generation error (CubeMX return code is 1).
Enable a verbose output or try to generate a code from the CubeMX itself.
ERROR    code generation error
```

ussserrr, 2020-02-13:
Run the app with `-v` argument to get the verbose output about the problem. Like this:
```
python stm32pio/app.py -v ...
```
Where you had installed STM32CubeMX? Because it looks like the default path specified in `settings.py` is not correct for your system. You can run `init` command first and then specify the correct path to the CubeMX in the created `stm32pio.ini` file.

kynguyen98, 2020-02-16:
Okay so first of all I'm running Manjaro which is Arch base system and I install STM32CubeMX through what is called AUR and the command was in 
```
/usr/bin/stm32cubemx
```
I do what you told me to correct the path to CubeMX created stm32pio.ini which is
```
[app]
java_cmd = java
platformio_cmd = platformio
cubemx_cmd = /usr/bin/stm32cubemx

[project]
cubemx_script_content = config load /home/john/STM32Cube/STM32F103C8/testing/testing.ioc
	generate code /home/john/STM32Cube/STM32F103C8/testing
	exit
platformio_ini_patch_content = [platformio]
	include_dir = Inc
	src_dir = Src
ioc_file = /home/john/STM32Cube/STM32F103C8/testing/testing.ioc
board = genericSTM32F103C8
```
And it gave me this
```
Error: Invalid or corrupt jarfile /usr/bin/stm32cubemx
```

opo2000tw, 2020-02-16:
1. Re-check java and stm32 installed path.

2.  This  complex command will execute, generate, & ... to default path on platformio  by stm32cubemx.
All of stm32cubemx command will be found in stm32 public "UM1718 User manual" and this line

```
cubemx_script_content': "config load $cubemx_ioc_full_filename\ngenerate code $project_path\nexit",
```

opo2000tw, 2020-02-16:
The other way to slove this is 
[1. Import Middleware & update default framework ](https://community.platformio.org/t/using-stm32cubemx-and-platformio/2611/57)
[2. Import Library to external lib by json](https://github.com/platformio/platform-ststm32/issues/217#issuecomment-488821662)

ussserrr, 2020-02-16:
@kynguyen98 OK, so I've been able to replicate your set-up. `/usr/bin/stm32cubemx` file isn't actually the `jar` file – that's why it's unable to start. This is just a script starting the actual CubeMX application. Don't remember exactly why I chose to invoke CubeMX as `java -jar ...`, probably for a portability on different systems. Actually, if you look inside `/usr/bin/stm32cubemx` file you see that what it's doing is exactly running `java -jar ...` command.

So the solution would be to locate the actual JAR file and specify it in `stm32pio.ini` config.

Also, you gave me a good idea to how to implement the CI finally (at least try to). Didn't know that it is somewhere possible to get CubeMX in "headless" mode, great that Arch providing such a feature. I've already tested the whole stm32pio application on Manjaro but installing CubeMX manually, downloading by hands.

kynguyen98, 2020-02-19:
Oh my god this actually worked!, thanks mate, right now I wanted to make the process of creating project more clearly
1. I opened STM32CubeMX and generate code 
2. stm32pio init -d ~/pathtomxproject -b genericSTM32F103C8 (without initialize the board it will give error)
3. edited the stm32pio.ini
4. ```python app.py new -d ~/pathtomxproject -b genericSTM32F103C8 -start-editor=code```
voila!
But every time I do that the mxproject didn't save to the platformio project workspace

closed 2020-02-19 by kynguyen98