ussserrr/stm32pio #11: qt.svg can't open file

kynguyen98, 2020-03-27:
Okay so I'm trying to run the new GUI feature, I install pyside on my machine but when I run the app it give me this
```
INFO <module> Starting stm32pio-gui...
qt.svg: Cannot open file '/home/john/Tetsing/stm32pio/stm32pio-gui/stm32pio-gui/icons/icon.svg', because: No such file or directory
qt.svg: Cannot open file '/home/john/Tetsing/stm32pio/stm32pio-gui/stm32pio-gui/icons/icon.svg', because: No such file or directory
QQmlApplicationEngine failed to load component
file:///home/john/Tetsing/stm32pio/stm32pio-gui/stm32pio-gui/main.qml: No such file or directory
Traceback (most recent call last):
  File "app.py", line 486, in <module>
    main_window = engine.rootObjects()[0]
IndexError: list index out of range
```
Even tho the icon is in the folder
```
[john@t440s stm32pio-gui]$ tree
.
├── app.py
├── icons
│   ├── add.svg
│   ├── icon.svg
│   ├── LICENSE
│   └── remove.svg
├── __init__.py
├── main.qml
└── README.md

1 directory, 8 files
```

ussserrr, 2020-04-15:
Currently it can be started only from the repository root point. So you need to navigate there and then run `python3 stm32pio-gui/app.py` to start.

Sorry for such a delayed reply, the GitHub notifications seems broken for me... Actually, I working on a new GUI version with many improvements, it should be released in a few days 😃

ussserrr, 2020-04-19:
There is [new GUI version](https://github.com/ussserrr/stm32pio/tree/master/stm32pio_gui) available now (1.21) which can be installed and ran both from `pip` and locally (from any directory).

kynguyen98, 2020-04-19:
Nice I’m gonna start working with it with my AUR. Is it okay for me to make this an AUR of my own?

ussserrr, 2020-04-19:
Sure, that's nice

closed 2020-04-19 by kynguyen98