ussserrr/stm32pio #12: Error: Unable to access jarfile C:/Program Files/STMicroelectronics/STM32Cube/STM32CubeMX/STM32CubeMX.exe

ChrisIdema, 2020-03-27:
I'm getting the following error:
`Error: Unable to access jarfile C:/Program Files/STMicroelectronics/STM32Cube/STM32CubeMX/STM32CubeMX.exe`
My installation is in the x86 folder:
`C:\Program Files (x86)\STMicroelectronics\STM32Cube\STM32CubeMX`
I have the latest version of python and the latest version of STM32CubeMX.

kynguyen98, 2020-03-27:
Have you tried navigate to settings.py and change 
```
Program Files
``` 
Program Files (x86)
```
in the stm32pio sub folder?

ChrisIdema, 2020-03-27:
I just did it and I got the same error message:
I changed `C:\Users\Chris\AppData\Roaming\Python\Python38\site-packages\stm32pio\settings.py`

kynguyen98, 2020-03-28:
Okay mate, I was able to replicate your system with windows installed on my machine here what I suggest you do 
* Open cmd with admin and type ```pip uinstall stm32pio```
* ```git clone  https://github.com/ussserrr/stm32pio.git ``` 
*  Right click on STM32CubeMx and select open file location then copy the **link** with STMCubeMx.exe like this 
```C:\Program Files\STMicroelectronics\STM32Cube\STM32CubeMX\STM32CubeMX.exe```
* Navigate to stm32pio folder in stm32pio repo you just git clone and open settings.py 
* Find the line 
```
pathlib.Path.home().joinpath("STM32CubeMX/STM32CubeMX") if my_os == 'Linux' else
            # Windows default: Program Files
            "C:/Program Files/STMicroelectronics/STM32Cube/STM32CubeMX/STM32CubeMX.exe" if my_os == 'Windows' else None
````
* Replace ```C:/Program Files/STMicroelectronics/STM32Cube/STM32CubeMX/STM32CubeMX.exe``` with the **link** you just copied 

For testing you just follow the guide on the repo README file

ChrisIdema, 2020-03-30:
Running `pip install .` produces the following error:
```
WARNING: pip is being invoked by an old script wrapper. This will fail in a future version of pip.
Please see https://github.com/pypa/pip/issues/5599 for advice on fixing the underlying issue.
To avoid this problem you can invoke Python with '-m pip' instead of running pip directly.
ERROR: Exception:
Traceback (most recent call last):
  File "C:\Users\Chris\AppData\Roaming\Python\Python38\site-packages\pip\_internal\cli\base_command.py", line 186, in _main
    status = self.run(options, args)
  File "C:\Users\Chris\AppData\Roaming\Python\Python38\site-packages\pip\_internal\commands\install.py", line 253, in run
    options.use_user_site = decide_user_install(
  File "C:\Users\Chris\AppData\Roaming\Python\Python38\site-packages\pip\_internal\commands\install.py", line 604, in decide_user_install
    if site_packages_writable(root=root_path, isolated=isolated_mode):
  File "C:\Users\Chris\AppData\Roaming\Python\Python38\site-packages\pip\_internal\commands\install.py", line 548, in site_packages_writable
    return all(
  File "C:\Users\Chris\AppData\Roaming\Python\Python38\site-packages\pip\_internal\commands\install.py", line 549, in <genexpr>
    test_writable_dir(d) for d in set(get_lib_location_guesses(**kwargs))
  File "C:\Users\Chris\AppData\Roaming\Python\Python38\site-packages\pip\_internal\utils\filesystem.py", line 140, in test_writable_dir
    return _test_writable_dir_win(path)
  File "C:\Users\Chris\AppData\Roaming\Python\Python38\site-packages\pip\_internal\utils\filesystem.py", line 153, in _test_writable_dir_win
    fd = os.open(file, os.O_RDWR | os.O_CREAT | os.O_EXCL)
PermissionError: [Errno 13] Permission denied: 'c:\\program files (x86)\\python38-32\\Lib\\site-packages\\accesstest_deleteme_fishfingers_custard_8ddh9t'
```

Running app.py results in the following error(it runs for a minute before the error message):
```
python app.py new -d "C:\Users\Chris\Documents\ee\software\stm32_serial\stm32_serial" -b nucleo_f446ze
INFO     starting to generate a code from the CubeMX .ioc file...
INFO     successful code generation
INFO     starting PlatformIO project initialization...
ERROR    FileNotFoundError: [WinError 2] The system cannot find the file specified
```

kynguyen98, 2020-03-30:
Okay mate there are 2 possible problems here:
1.You forgot to install platformio client so the last app.py output 
```ERROR    FileNotFoundError: [WinError 2] The system cannot find the file specified``` so I **suggest** you do ```pip install platformio```  
2. When you ```pip install .``` it spit out
```
WARNING: pip is being invoked by an old script wrapper. This will fail in a future version of pip.
Please see https://github.com/pypa/pip/issues/5599 for advice on fixing the underlying issue.
To avoid this problem you can invoke Python with '-m pip' instead of running pip directly.
``` 
My windows work fine with just ```pip install .``` Windows is weird tho but I suggest you
 ```pip install --upgrade pip``` to update pip  then ```python -m pip install .``` to install again

Oh and if you manage to solve your problem try ```stm32pio```  anywhere instead of ```python app.py``` in the stm32pio folder

ussserrr, 2020-04-23:
No response -> close the issue

closed 2020-04-23 by ussserrr

dzid26, 2020-08-03:
> 1.You forgot to install platformio client so the last app.py output
> `ERROR FileNotFoundError: [WinError 2] The system cannot find the file specified` so I **suggest** you do `pip install platformio`
I had the same error and this worked!

It would be useful if the error message was saying what to install (`pip install platformio`).

creatormir, 2021-02-20:
I did on instructions, result:

>stm32pio generate
INFO     starting to generate a code from the CubeMX .ioc file...
Return code is 1
Output:
java.lang.UnsupportedClassVersionError: com/st/microxplorer/maingui/STM32CubeMX : Unsupported major.minor version 52.0
        at java.lang.ClassLoader.defineClass1(Native Method)
        at java.lang.ClassLoader.defineClass(Unknown Source)
        at java.security.SecureClassLoader.defineClass(Unknown Source)
        at java.net.URLClassLoader.defineClass(Unknown Source)
        at java.net.URLClassLoader.access$100(Unknown Source)
        at java.net.URLClassLoader$1.run(Unknown Source)
        at java.net.URLClassLoader$1.run(Unknown Source)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.net.URLClassLoader.findClass(Unknown Source)
        at java.lang.ClassLoader.loadClass(Unknown Source)
        at sun.misc.Launcher$AppClassLoader.loadClass(Unknown Source)
        at java.lang.ClassLoader.loadClass(Unknown Source)
        at sun.launcher.LauncherHelper.checkAndLoadMain(Unknown Source)

ERROR    code generation error
INFO     Traceback has been saved to the stm32pio.ini. It will be cleared on the next successfulrun

ussserrr, 2021-02-20:
Most likely you have installed incompatible version of Java (which is required for CubeMX). If we looking in its PDF docs:
> 64-bit version mandatory; 32-bit version not supported
> Version 11 is supported
> Versions 7, 9, 10, 12 and upper are not supported