ussserrr/stm32pio #13: Can not generate code with I2C support

romychs, 2020-04-14:
Have success with generation code for bluePill board with STM32F103C8T6, but after turning ON I2C 1 in Stm32CubeMX I am get error:
```
romych@home_pc ~/projects/TEnco % stm32pio generate                                                                             
INFO     starting to generate a code from the CubeMX .ioc file...
ERROR    2020-04-14 21:25:39,422 [ERROR] IPConfigManager:1272 - ERROR: the RefConfig I2C_Analogue is not available in your config.xml file
2020-04-14 21:25:39,422 [ERROR] IPConfigManager:1272 - ERROR: the RefConfig I2C_DigitalFolter is not available in your config.xml file
ERROR    Exception: code generation error
```
Env:
LinuxMint 19, fresh platformio v 4.3.1, fresh Stm32CubeMX v 5.6.1, VS Code v1.44

[TEnco.zip](https://github.com/ussserrr/stm32pio/files/4477119/TEnco.zip)

romychs, 2020-04-14:
I think, it is CubeMX issue, not stm32pio...

ussserrr, 2020-04-15:
Yea, I think so, too. Are you able to generate the code from the same configuration using the ordinary GUI version of CubeMX?

kynguyen98, 2020-04-15:
> I think, it is CubeMX issue, not stm32pio...

I don't think so mate mine was perfectly fine 
```
[john@t440s testing]$ stm32pio new -b genericSTM32F103C8
INFO     starting to generate a code from the CubeMX .ioc file...
INFO     successful code generation
INFO     starting PlatformIO project initialization...
INFO     successful PlatformIO project initialization
INFO     'platformio.ini' has been patched
INFO     exiting...
```
```
[john@t440s testing]$ tree
.
├── Inc
│   ├── gpio.h
│   ├── i2c.h
│   ├── main.h
│   ├── stm32f1xx_hal_conf.h
│   └── stm32f1xx_it.h
├── lib
│   └── README
├── platformio.ini
├── Src
│   ├── gpio.c
│   ├── i2c.c
│   ├── main.c
│   ├── stm32f1xx_hal_msp.c
│   ├── stm32f1xx_it.c
│   └── system_stm32f1xx.c
├── stm32pio.ini
├── test
│   └── README
├── testing.gpdsc
└── testing.ioc
```

ussserrr, 2020-04-19:
OK, I've been able to replicate this issue. This is indeed the stm32pio bug, but... CubeMX doesn't return any error code when it encounters the problem and just always returns 0. So there is a little bit hacky solution to detect when the real error has occurred and when not. Currently, I look for the `ERROR` string to appear in the logs to mark the action as failed. Apparently, in your case ERROR is just a part of a normal process so this logic has failed. stm32pio sees it and interrupts the execution. Need to find out the other way of distinguishing the problems, probably the final `OK` string is a good indicator. Need to be fixed.

ussserrr, 2020-05-05:
Parsing behavior of CubeMX output is changed in the new version, close.

closed 2020-05-05 by ussserrr

romychs, 2020-05-05:
Thanks for the new release!