ussserrr/stm32pio #21: `generate` command fails if `.ioc` file has space in file name or directory

RossBencina, 2020-12-10:
I have the following project directory:

  `C:\Users\Desktop\Ross\STM32Learning`

containing only the `.ioc` file named `STM32 Learning.ioc` (there is a space between "STM32" and "Learning".

I ran the following commands:

>C:\Python37\Scripts\stm32pio.exe init
>C:\Python37\Scripts\stm32pio.exe generate

Then the following error was displayed.

[image]
 
The problem goes away if I rename the `.ioc` file without a space.

I get a similar error if the working directory has a space in the directory name, e.g. 

  `C:\Users\Desktop\Ross\STM32 Learning`

In the latter case, when running `generate` a new directory named  `C:\Users\Desktop\Ross\STM32` is created (e.g. the tool is ignoring the directory name from the space onwards).

I don't know the cause of the problem but possibly there are some missing quotation marks somewhere.

Please describe the bug and supplement the additional information:
 - Windows 10, Python 3.7.3
 - content of the config file `stm32pio.ini` => file does not exist.
 - your project folder tree (see above)

Verbose command output:

```
C:\Users\Ross\Desktop\STM32Learning>C:\Python37\Scripts\stm32pio.exe -v generate
DEBUG    setup_logging              debug logging enabled
DEBUG    _load_config               searching for stm32pio.ini...
DEBUG    _find_ioc_file             using 'STM32 Learning.ioc' file from the INI config
DEBUG    __init__                   resolved config:
========== app ==========
('java_cmd', 'java')
('platformio_cmd', 'platformio')
('cubemx_cmd', 'C:/Program Files/STMicroelectronics/STM32Cube/STM32CubeMX/STM32CubeMX.exe')

========== project ==========
('cubemx_script_content', 'config load ${ioc_file_absolute_path}\ngenerate code ${project_dir_absolute_path}\nexit')
('platformio_ini_patch_content', '[platformio]\ninclude_dir = Inc\nsrc_dir = Src')
('board', '')
('ioc_file', 'STM32 Learning.ioc')

INFO     generate_code              starting to generate a code from the CubeMX .ioc file...
log4j user configuration file not found: C:\Users\Ross/.stm32cubemx/log4j.properties
Configure log4j with default settings from jar:file:/C:/Program%20Files/STMicroelectronics/STM32Cube/STM32CubeMX/STM32CubeMX.exe!/log4j/log4j.stm32cubemx.properties
2020-12-10 21:15:41,923 [INFO] MicroXplorer:77 - [MX] MX Start == 2246992563080500
2020-12-10 21:15:41,926 [INFO] MicroXplorer:565 - Detected Java Version = 11.0.9.1
2020-12-10 21:15:41,933 [INFO] ApplicationProperties:181 - Using Application install path: C:\Program Files\STMicroelectronics\STM32Cube\STM32CubeMX
2020-12-10 21:15:41,949 [INFO] DbMcusXml:71 - Set database path to: C:\Program Files\STMicroelectronics\STM32Cube\STM32CubeMX\\db\/mcu/
2020-12-10 21:15:41,956 [INFO] DbBoardsPdsc:57 - Set plugin database path to: C:\Program Files\STMicroelectronics\STM32Cube\STM32CubeMX\\db\/plugins/boardmanager/
2020-12-10 21:15:41,957 [INFO] DbMcus:259 - Set plugin images path to: C:\Program Files\STMicroelectronics\STM32Cube\STM32CubeMX\\db\/plugins/mcufinder/images/
2020-12-10 21:15:41,957 [INFO] DbBoards:265 - Set plugin images path to: C:\Program Files\STMicroelectronics\STM32Cube\STM32CubeMX\\db\/plugins/mcufinder/images/
2020-12-10 21:15:41,960 [INFO] DbExamples:329 - Set plugin images path to: C:\Program Files\STMicroelectronics\STM32Cube\STM32CubeMX\\db\/plugins/mcufinder/images/
2020-12-10 21:15:41,961 [INFO] McuFinderGlobals:40 - Set McuFinder mode to 1 (CubeMX integrated)
2020-12-10 21:15:41,969 [INFO] DbMcusDocs:112 - Set database path to: C:\Users\Ross\.stmcufinder\plugins\mcufinder//mcu/
2020-12-10 21:15:41,971 [INFO] DbMcusJson:63 - Set database path to: C:\Users\Ross\.stmcufinder\plugins\mcufinder//mcu/
2020-12-10 21:15:41,972 [INFO] DbBoardsDocs:112 - Set database path to: C:\Users\Ross\.stmcufinder\plugins\mcufinder//mcu/
2020-12-10 21:15:41,974 [INFO] DbBoardsJson:56 - Set database path to: C:\Users\Ross\.stmcufinder\plugins\mcufinder//mcu/
2020-12-10 21:15:42,245 [INFO] CrossReferenceDbSqlite:196 - Set database path to: C:\Users\Ross\.stmcufinder\plugins\mcufinder//mcu/cs/
2020-12-10 21:15:42,249 [INFO] DbExamplesSqlite:699 - Set database path to: C:\Users\Ross\.stmcufinder\plugins\mcufinder//mcu/
2020-12-10 21:15:42,474 [INFO] RulesReader:52 - Compatibility file has been processed (228 Rules)
2020-12-10 21:15:42,552 [INFO] STM32CubeMX:31 - Starting STM32CubeMX
2020-12-10 21:15:43,366 [INFO] MainPanel:190 - HeapMemory: 536870912
[... 152 lines cut ...]
2020-12-10 21:15:52,084 [WARN] ConditionMgr:437 - getConditionDescription Invalid condition id : Sigfox Third Party LIB Condition cause : null
2020-12-10 21:15:52,090 [WARN] ConditionMgr:1040 - genDependencies : Invalid condition id : SIGFOX STM32 LIB Condition cause : null
2020-12-10 21:15:52,090 [WARN] ConditionMgr:1040 - genDependencies : Invalid condition id : Sigfox Third Party LIB Condition cause : null
2020-12-10 21:15:52,091 [WARN] ConditionMgr:1040 - genDependencies : Invalid condition id : Sigfox Third Party LIB Condition cause : null
2020-12-10 21:15:52,092 [WARN] ConditionMgr:1040 - genDependencies : Invalid condition id : SIGFOX STM32 LIB Condition cause : null
2020-12-10 21:15:52,092 [WARN] ConditionMgr:1040 - genDependencies : Invalid condition id : SIGFOX STM32 LIB Condition cause : null
2020-12-10 21:15:52,092 [WARN] ConditionMgr:1040 - genDependencies : Invalid condition id : Sigfox Third Party LIB Condition cause : null
2020-12-10 21:15:52,476 [INFO] ThirdParty:887 - Integrity check success = true
2020-12-10 21:15:52,477 [INFO] IntegrityCheckThread:75 - exiting critical section [integrity check]
2020-12-10 21:15:52,477 [INFO] IntegrityCheckThread:78 - End integrity checks thread
2020-12-10 21:15:52,477 [INFO] ThirdParty:785 - entering critical section [close project]
2020-12-10 21:15:52,478 [INFO] ThirdParty:796 - exiting critical section [close project]
2020-12-10 21:15:52,481 [INFO] PinOutPanel:1434 - setPackage(No Configuration,No Configuration)
2020-12-10 21:15:52,566 [INFO] MainFrame:167 - [MX] HOME_PANEL == 2247003207398000
config load C:\Users\Ross\Desktop\STM32Learning\STM32 Learning.ioc
2020-12-10 21:15:52,586 [INFO] HomeAdPanel:149 - scale :0.9666048237476809
KO
generate code C:\Users\Ross\Desktop\STM32Learning
2020-12-10 21:16:00,557 [INFO] CodeGenerator:639 - code generatio: config db path: C:\Program Files\STMicroelectronics\STM32Cube\STM32CubeMX\\db\
Exception in code generation java.lang.StringIndexOutOfBoundsException: begin 10, end 11, length 0
KO
exit
Bye bye
WARNING  generate_code              Undefined result from the CubeMX (neither error or success symptoms were found in the logs). Keep going but there might be an error
```

ussserrr, 2020-12-22:
I can confirm it (also not only for Windows).

> I don't know the cause of the problem but possibly there are some missing quotation marks somewhere.

Yep, it's exactly like that. Turns out I manage paths correctly in Python but forgot to provide them properly to the CubeMX. Different software struggling with spaced names for years now so I made it a habit to never assign such names to my files/folders 😄  and so never test it either.

P.S. Probably I should reach the GitHub support again, I still don't receive any issues notifications at all...

ussserrr, 2021-01-09:
Fixed in 2.1.0

closed 2021-01-09 by ussserrr