ussserrr/stm32pio #22: `init` followed by `new` command fails on Windows

RossBencina, 2020-12-10:
Here's what I did:

- run `init`
- edit the `board = ` field in `stm32pio.ini`
- run 'new`

Error: ` FileNotFoundError: [WinError 2] The system cannot find the file specified
Traceback (most recent call last):`

Full log below.

Windows 10, Python 3.7.3

To begin, directory only contains `STM32Learning.ioc`

stm32pio.ini after editing
```
[app]
java_cmd = java
platformio_cmd = platformio
cubemx_cmd = C:/Program Files/STMicroelectronics/STM32Cube/STM32CubeMX/STM32CubeMX.exe

[project]
cubemx_script_content = config load ${ioc_file_absolute_path}
	generate code ${project_dir_absolute_path}
	exit
	
platformio_ini_patch_content = [platformio]
	include_dir = Inc
	src_dir = Src
	
board = blackpill_f411ce
ioc_file = STM32Learning.ioc

```

Full log:

```
C:\Users\Ross\Desktop\STM32Learning>ls
STM32Learning.ioc

C:\Users\Ross\Desktop\STM32Learning>C:\Python37\Scripts\stm32pio.exe -v init
DEBUG    setup_logging              debug logging enabled
DEBUG    _load_config               searching for stm32pio.ini...
DEBUG    _find_ioc_file             searching for any .ioc file...
DEBUG    _find_ioc_file             STM32Learning.ioc is selected
DEBUG    __init__                   resolved config:
========== app ==========
('java_cmd', 'java')
('platformio_cmd', 'platformio')
('cubemx_cmd', 'C:/Program Files/STMicroelectronics/STM32Cube/STM32CubeMX/STM32CubeMX.exe')

========== project ==========
('cubemx_script_content', 'config load ${ioc_file_absolute_path}\ngenerate code ${project_dir_absolute_path}\nexit\n')
('platformio_ini_patch_content', '[platformio]\ninclude_dir = Inc\nsrc_dir = Src\n')
('board', '')
('ioc_file', 'STM32Learning.ioc')

WARNING  main                       PlatformIO board identifier is not specified, it will be needed on PlatformIO project creation. Type 'pio boards' or go to https://platformio.org to find an appropriate identifier
INFO     main                       project has been initialized. You can now edit stm32pio.ini config file
DEBUG    _save_config               stm32pio.ini config file has been saved

C:\Users\Ross\Desktop\STM32Learning>notepad stm32pio.ini

C:\Users\Ross\Desktop\STM32Learning>C:\Python37\Scripts\stm32pio.exe -v new
DEBUG    setup_logging              debug logging enabled
DEBUG    _load_config               searching for stm32pio.ini...
DEBUG    _find_ioc_file             using 'STM32Learning.ioc' file from the INI config
DEBUG    __init__                   resolved config:
[... 402 lines cut ...]
2020-12-10 21:46:45,344 [INFO] CodeEngine:229 - oldGeneratedFile, C:\Users\Ross\Desktop\STM32Learning\Src\main.c_save
2020-12-10 21:46:45,522 [INFO] CodeEngine:247 - Generated code: C:\Users\Ross\Desktop\STM32Learning\Src\main.c
Code succesfully generated under C:\Users\Ross\Desktop\STM32Learning
OK
exit
Bye bye
INFO     generate_code              successful code generation
INFO     pio_init                   starting PlatformIO project initialization...
ERROR    log_current_exception      FileNotFoundError: [WinError 2] The system cannot find the file specified
Traceback (most recent call last):
  File "c:\python37\lib\site-packages\stm32pio\cli\app.py", line 179, in main
    project.pio_init()
  File "c:\python37\lib\site-packages\stm32pio\core\lib.py", line 473, in pio_init
    result = subprocess.run(command_arr, encoding='utf-8', stdout=subprocess.PIPE, stderr=subprocess.STDOUT)
  File "c:\python37\lib\subprocess.py", line 472, in run
    with Popen(*popenargs, **kwargs) as process:
  File "c:\python37\lib\subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "c:\python37\lib\subprocess.py", line 1178, in _execute_child
    startupinfo)

DEBUG    _save_config               stm32pio.ini config file has been saved

C:\Users\Ross\Desktop\STM32Learning>
```

ussserrr, 2020-12-22:
Do you able to run the PlatformIO as `platformio` command in command line? The error seems to look like that... Actually, I'm currently working on the "environment validation" feature that will help users to check all tools used by stm32pio to be present in the system.

RossBencina, 2020-12-28:
> Do you able to run the PlatformIO as platformio command in command line? 

Not as a single `platformio` command:

```
C:\Users\Ross\Desktop\STM32Learning>platformio
'platformio' is not recognized as an internal or external command, operable program or batch file.
```

I run it like this `py -m platformio`:

```
C:\Users\Ross\Desktop\STM32Learning>py -m platformio
Usage: __main__.py [OPTIONS] COMMAND [ARGS]...

Options:
  --version          Show the version and exit.
  -f, --force        DEPRECATE
  -c, --caller TEXT  Caller ID (service)
  --no-ansi          Do not print ANSI control characters
  -h, --help         Show this message and exit.

Commands:
  access    Manage resource access
  account   Manage PlatformIO account
  boards    Embedded board explorer
  check     Static code analysis
  ci        Continuous integration
  debug     Unified debugger
  device    Device manager & serial/socket monitor
  home      UI to manage PlatformIO
  lib       Library manager
  org       Manage organizations
  package   Package manager
  platform  Platform manager
  project   Project manager
  remote    Remote development
  run       Run project targets (build, upload, clean, etc.)
  settings  Manage system settings
  system    Miscellaneous system commands
  team      Manage organization teams
  test      Unit testing
  update    Update installed platforms, packages and libraries
  upgrade   Upgrade PlatformIO to the latest version
```

RossBencina, 2020-12-28:
After adding the `Python37` and `Python37\Scripts` directories to my `PATH` *before* the `Python27` (Python 2) paths I have been able to make progress and no longer get the error described above. Thank you for your help.

ussserrr, 2021-01-09:
Starting from 2.1.0 (#25), one can use validation feature to verify their tools' installation for the particular project (its own config, or default one). Like this:
```
$ stm32pio validate -d tests/fixtures/nucleo_f031k6
WARNING  config file not found. Validation will be performed against the runtime configuration
INFO     checking 'java_cmd'...
INFO     checking 'cubemx_cmd'...
INFO     checking 'platformio_cmd'...
[error]  java_cmd  
[error]  cubemx_cmd
[error]  platformio_cmd

Tools output:

java_cmd
    The operation couldn’t be completed. Unable to locate a Java Runtime.
Please visit http://www.java.com for information on installing Java.

cubemx_cmd
    The operation couldn’t be completed. Unable to locate a Java Runtime.
Please visit http://www.java.com for information on installing Java.

platformio_cmd
    [Errno 2] No such file or directory: 'platformio'
```

Also, I've planned to add support for the `python -m platformio` option. Closing this for now, thanks for the feedback.

closed 2021-01-09 by ussserrr