ussserrr/stm32pio #24: MacOS Big Sur Error

ehtnevets, 2021-01-09:
First, stm32pio generates stm32pio.ini  with wrong cubemx_cmd directory. Solution is not bad, just modify to the correct path. But this one,

INFO     starting to generate a code from the CubeMX .ioc file...
INFO     successful code generation
INFO     starting PlatformIO project initialization...
ERROR    FileNotFoundError: [Errno 2] No such file or directory: 'platformio': 'platformio'

Not sure what's looking for. A platformio command line? I installed platformio as vscode extension.

ussserrr, 2021-01-09:
PlatformIO IDE that you use inside the VSCode is built on top of the PlatformIO Core – a command-line version of the PlatformIO. You can activate it without any additional installations. See [its docs](https://docs.platformio.org/en/latest/core/installation.html#piocore-install-shell-commands). stm32pio is using it too, to trigger PlatformIO-related operations.

Also, can you please tell me your new (edited) `cubemx_cmd` value? How did you install the CubeMX? Currently, I just unpack ZIP archive and run the executable, maybe there is a straightforward way to install it that I missed.

ehtnevets, 2021-01-09:
Thank you. 

I modified cubemx_cmd path to /Applications/STMicroelectronics/STM32CubeMX.app/Contents/Resources/STM32CubeMX.

ussserrr, 2021-01-09:
OK, I see it now. There are actually 2 types of ZIP archives that ST distributes, and their content is different too, so I was tricked by that. I've already changed the default value to the path you supply.

Closed by #25.

closed 2021-01-09 by ussserrr

ehtnevets, 2021-01-09:
I know you've closed the this topic but I was still getting this error,
INFO     starting to generate a code from the CubeMX .ioc file...
INFO     successful code generation
INFO     starting PlatformIO project initialization...
Return code is 1. Output:

Traceback (most recent call last):
  File "/Users/SThe/.platformio/penv/bin/platformio", line 5, in <module>
    from platformio.__main__ import main
ModuleNotFoundError: No module named 'platformio'

Entering venv with python3 helps and gets me through stm32pio successfully. I'm not sure if this tool support project generated by TouchGFX Designer but I guess I have to tweek platformio.ini include path.

ussserrr, 2021-01-09:
First of all, make sure you can start the PlatformIO CLI manually before launching the stm32pio. It uses the exact command `platformio` just like you would type it yourself, and it has no sense trying to start the app before it can be successfully run on its own. I don't know your particular set-up to give you some specific advices, though. Glad you at least were been able to use it through the venv tool.

I hadn't have any experience with the TouchGFX kit so, unfortunately, no particular guides here either. I can suggest you to reach the PlatformIO docs since it will be responsible for the eventual build of your project. Yeah, potentially some `platformio.ini` tweaks can help to accomplish the goal. Those parameters can also probably be placed under the `platformio_ini_patch_content` key of the `stm32pio.ini` config.