ussserrr/stm32pio #28: Release version won't install but could install via git clone the repo directly

kynguyen98, 2021-04-16:
## Machine Information
 - Arch Linux, Python3 version 3.9.3
- Installed package: ```python-setuptools``` ```python-setuptools-csm```
## Problem 
- When I try to install the release archive v2.1.0 i.e the ```stm32pio-2.1.0.tar.gz```, I can't no longer install it like this anymore ```python setup.py install ```. It gave me this error 
```
LookupError: setuptools-scm was unable to detect version for '/home/alex/Downloads/stm32pio/src/stm32pio-2.1.0'.

Make sure you're either building from a fully intact git repository or PyPI tarballs. Most other sources (such as GitHub's tarballs, a git checkout without the .git folder) don't contain the necessary metadata and will not work.

For example, if you're using pip, instead of https://github.com/user/proj/archive/master.zip use git+https://github.com/user/proj.git#egg=proj
```
But I was able to install it by cloning the repo directly like this
```
git clone https://github.com/ussserrr/stm32pio.git
cd stm32pio
python setup.py install --user 
sudo ln -s ~/.local/bin/stm32pio /usr/bin/stm32pio
```

ussserrr, 2021-04-16:
Did you try the recommended command from the [README](https://github.com/ussserrr/stm32pio#installation)?
```bash
$ tar -xf stm32pio-2.1.0.tar.gz
$ cd stm32pio-2.1.0
$ pip install .
```

Btw, these `tar.gz` and `zip` archives are auto-generated by GitHub every time I issue a new release, they are just packed snapshots of the repository root and not some kind of built wheels or something like this.

kynguyen98, 2021-04-17:
Nope mate, I'm still getting the same error but in different variant 
```
    LookupError: setuptools-scm was unable to detect version for '/tmp/pip-req-build-nscicne9'.
    
    Make sure you're either building from a fully intact git repository or PyPI tarballs. Most other sources (such as GitHub's tarballs, a git checkout without the .git folder) don't contain the necessary metadata and will not work.
    
    For example, if you're using pip, instead of https://github.com/user/proj/archive/master.zip use git+https://github.com/user/proj.git#egg=proj
    ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python /usr/lib/python3.9/site-packages/pep517/in_process/_in_process.py prepare_metadata_for_build_wheel /tmp/tmpayzup4cz Check the logs for full command output.
```
I just checked out the the repo's version, up to ```v1.30```. I was able to successfully install with ```pip install .``` or ```python setup.py install```, version ```2.0.0``` and up failed to install

ussserrr, 2021-04-17:
OK, I got it. Starting from v2.0.0 the app version is not hardcoded or listed anywhere in the project files and specified only as a git repository tag. That's what the `setuptools-scm` package is used for after all. So, in order to install from source you need to either clone the repo (you've already figured it out) or, if you decided to download the `tar.gz`, just use it in "portable" (not installed) manner (or create your own git repository inside the folder).

kynguyen98, 2021-04-17:
Thanks, mate. I think I'll just gonna clone the repo directly from now with the latest release of course. AUR package updated. Closing this issue now.

closed 2021-04-17 by kynguyen98