ussserrr/stm32pio #30: platformio_cmd not found

Lutz1967, 2021-05-15:
Hi,

Using CLI on MacOS10.13.6

Problem when running validate:

INFO     checking 'java_cmd'...
INFO     checking 'cubemx_cmd'...
INFO     checking 'platformio_cmd'...
[   ok]  java_cmd  
[   ok]  cubemx_cmd
[error]  platformio_cmd

how do I fix that? I use PlatformIO all the time in VSCode.
Thanks!

ussserrr, 2021-05-15:
How do you install the PlatformIO? If it was PlatformIO IDE then you probably need to add _shell extensions_ so you (and stm32pio) can launch PlatformIO through the CLI. See [there](https://docs.platformio.org/en/latest/core/installation.html#piocore-install-shell-commands) for more information.

Lutz1967, 2021-05-15:
Thanks - solved.

Lutz1967, 2021-05-15:
Thanks for the quick answer.

closed 2021-05-15 by Lutz1967