ussserrr/stm32pio #31: Generate into another folder

cooked, 2021-07-06:
Hi,
is it possible to generate into a different folder than the one containing the .ioc file?
I'd like to keep many .ioc files in a subfolder, then run and build into the root folder of my project

ussserrr, 2021-07-07:
It is currently not available. I think you can mimic such a feature, for example by some shell script automatically copying an .ioc file from your main repository to the project folder and therefore always keeping it “in sync” (and passing all other arguments to stm32pio). Overall, it is probably a useful thing to implement in the future.

cooked, 2021-07-07:
@ussserrr thanks for the feedback, indeed I'm thinking of wrapping some tooling functions like this in an external script