ussserrr/stm32pio #78: Middleware support

ussserrr, 2021-08-08:
FreeRTOS, etc.

cooked, 2021-09-16:
Hi @ussserrr I'm currently working on a project that uses USB CDC middleware. I stumbled on the related posts on platformio forum and then on this issue. 
At the moment I'm testing the workaround described there to let CubeMX generate the middleware USB code, then copy the Middleware folders as well as the Application file over to the platformio project
stm32pio middleware support will streamline and make the whole process way neater
any estimate of when this will be implemented ?

ussserrr, 2021-09-16:
Hello. I think middlewares are too broad to cover all of them. If you can describe needed workflow as detailed as possible, provide an example, then I will try to implement this feature. For example:

- What options should CubeMX `.ioc` project have?
- How `platformio.ini` config should be tweaked to be able to compile?

Maybe you can list some PlatformIO cases where it is used to build such a code generated by the CubeMX. Specify a target MCU/family, kind of middleware, etc. Currently, I don't have much free time, though, so I can't tell you exactly how long this will take.

cooked, 2021-12-05:
sorry for the late late response. I found quite often in need of the USB functionality so I would start with just that one. I think this is quite general use case on fro stm32f1 and stm32f4 family of mcus and dev boarads. 
Trying to fish out  the info from a project I have now:

- .ioc options (not sure if this is all of them though)
USB_DEVICE.CLASS_NAME_FS=CDC
USB_DEVICE.IPParameters=VirtualMode,VirtualModeFS,CLASS_NAME_FS
USB_DEVICE.VirtualMode=Cdc
USB_DEVICE.VirtualModeFS=Cdc_FS
USB_OTG_FS.IPParameters=VirtualMode
USB_OTG_FS.VirtualMode=Device_Only
VP_USB_DEVICE_VS_USB_DEVICE_CDC_FS.Mode=CDC_FS
VP_USB_DEVICE_VS_USB_DEVICE_CDC_FS.Signal=USB_DEVICE_VS_USB_DEVICE_CDC_FS

- generated code I currently put it in a structure like one below, but possibly that can be put in lib where it gets picked up by platformio automatically
[image]

- platformio.ini shall have the following added (in case of the location above)
lib_extra_dirs = 
	Middlewares/ST/STM32_USB_Device_Library

ussserrr, 2022-01-09:
@cooked OK, I'll take a look at it. Though, it seems like the only thing stm32pio should do here is to add the `lib_extra_dirs` option to the `platformio.ini`, not necessarily a complex task...