ussserrr/stm32pio #103: allow specifying alternate names for "Src" and "Inc" in stm32pio.ini

rickyelopez, 2021-09-26:
Hello! Love this project, very useful

In my `stm32pio.ini` files I'm using `platformio_ini_patch_content` to set the `include_dir` and `src_dir`  keys to alternate names. However, there doesn't seem to be a way to propagate that change to the code generation, so whenever I run `stm32pio generate`, I then need to deal with renaming the directories which are still created as `Src` and `Inc`.

It would be really great if we could also specify the output directory names from calling `stm32pio generate`. Maybe this already exists and I'm just missing it?

ussserrr, 2021-09-27:
Hi! What you're asking for is to tell the CubeMX where to place its generated sources but there is no such option. So it can only be done by renaming already created folders. But why you want this feature, what it will be useful for?

rickyelopez, 2021-09-27:
Ah, that's unfortunate. It isn't strictly necessary, just would have been nice to have the generated files go directly into the directories that I want without having to move them manually. Not the end of the world though, thanks anyway!

closed 2021-09-27 by rickyelopez