ussserrr/stm32pio #114: Migrate "GPDSC" to "STM32CubeIDE" for default code-generation

ussserrr, 2022-02-06:
### Discussed in https://github.com/ussserrr/stm32pio/discussions/113

<div type='discussions-op-text'>

<sup>Originally posted by **FrankBau** February  5, 2022</sup>
<b>Warning: GPDSC generation will be soon discontinued</b><br>
With STM32CubeMX 6.4.0 (Nov. 2021) I get a warning on code generation:

  - The support of the “Tool chain / IDE” you have selected will be soon discontinued in further STM32CubeMX version.   If you want to benefit from further STM32CubeMX evolutions for your project,   we advise you to target STM32CubeIDE or other IDE.

What can we do?

KnarfB
</div>

vanbwodonk, 2022-09-29:
FYI, GPDSC generation already discontinued

mron, 2024-06-13:
Please update the readme.md! The default toolchain seems to be EWARM. When I use stm32pio I get "It is recommended to use value "Other Toolchains (GPDSC)". This implies to me that, since "Other.." is no longer available, EWARM is OK. It should be "STM32CubeIDE".  On an M1 Mac, UART serial out fails.

dufguix, 2025-02-17:
Feb 2025: Available options in CubeMX are : EWARM, MDK-ARM, STM32CubeIDE, Makefile, CMake.

What are we supposed to use instead of "Other Toolchains (GPDSC)" ?

ussserrr, 2025-02-19:
I think something from STM32CubeIDE, Makefile, CMake should be fine. AFAIR PlatformiIO STM32 projects utilizes basically plain C project structure, without any fancy custom hacks.